Timeular/noble-mac #6: Symbol not found: _napi_create_error

SamDecrock, 2018-10-12:
Hi, when using noble-mac as a drop-in replacement for noble, I get the following error:

```js
module.js:597
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: dlopen(/Users/samdecrock/projects/blefun/node_modules/noble-mac/native/noble_mac.node, 1): Symbol not found: _napi_create_error
  Referenced from: /Users/samdecrock/projects/blefun/node_modules/noble-mac/native/noble_mac.node
  Expected in: flat namespace
 in /Users/samdecrock/projects/blefun/node_modules/noble-mac/native/noble_mac.node
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/Users/samdecrock/projects/blefun/node_modules/noble-mac/lib/binding.js:4:48)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)

```

geovie, 2018-10-12:
Which node version are you using? You need to add the --napi-modules argument (node --napi-modules <your script>

SamDecrock, 2018-10-12:
I was using node v6.9.1, and the --napi-modules argument didn't work.
So I upgraded to the latest version (v8.12.0), now it works without the --napi-modules

Thanks

closed 2018-10-12 by SamDecrock