Timeular/noble-mac #7: Not Discovering all Peripherals

jaret32, 2018-10-29:
I have an Adafruit Feather 32u4 Bluefruit LE board that I am trying to connect to.  Before upgrading to MacOS Mojave and using the original noble branch, the peripheral could be discovered.  After upgrading, noble was unable to discover any peripherals.  Since then I have tried using this repository instead, and it does detect some peripherals, but not the board I am trying to connect to.  I am not sure what the issue is, please let me know if you have any suggestions.

s524797336, 2018-10-30:
Give this a try see if it works
```
npm install @s524797336/noble-mac@0.0.7
```

jaret32, 2018-10-30:
Yes, it worked.  Thank you!  Can I ask what exactly that does to make it work?

geovie, 2018-11-01:
@s524797336 Thanks for investigating this issue!
Unfortunately I can't reproduce the issue (not that many ble devices around...)
How many devices do you get in a single scan?

Could you check if all devices are delivered to the callback by logging the `uuid` of the devices (`printf("%s\n", uuid.c_str());`) in the `Emit::Scan` method (this line https://github.com/Timeular/noble-mac/blob/master/src/callbacks.cc#L89)

akinyema, 2018-11-01:
Hi,
 using that i get Segmentation fault: 11 error at random. Sometimes it works and sometimes it doesn't, Is there a way to stop this?

s524797336, 2018-11-02:
@geovie  
I don't known how to run C code, even you don't have many devices, you can start a thread to simulation a lot of callback and see if some of them are not emit or delay a lot of time to emit.
I'm not sure this issue is problem, but reduce emit dose output what i want.

geovie, 2018-11-06:
I tested it and the simulation works for me (the library we are use has a [testcase](https://github.com/mika-fischer/napi-thread-safe-callback/blob/master/test/tests.cpp#L150) for this problem).
If your peripherals are advertising their service uuids you could filter by that.
Which node version are you using?

s524797336, 2018-11-07:
@geovie
node 10.10.0

s524797336, 2018-11-09:
@geovie 
I'm in hospital, i'll do that when i got the chance.

ReinierH, 2018-12-05:
@geovie 
That doesn't fix it

geovie, 2018-12-05:
@ReinierH Can you try the [test/not-discovered](https://github.com/Timeular/noble-mac/tree/test/not-discovered) branch and post the results here?

ReinierH, 2018-12-13:
@geovie 
Sorry for not getting back to you earlier.
I tested it and It now can discover peripherals but the localName stays empty. 
I'm running it on nodejs v8.11.3

geovie, 2018-12-13:
@ReinierH Have you tried the latest version (0.0.3)? It was released yesterday and had some changes regarding advertiment handling...

ReinierH, 2018-12-14:
@geovie 
Just tried discovering devices with 0.0.3, it seems that some of the properties in "advertisement" are missing.

I've included the console.log of a peripheral below
`     _events:
      { warning: [Function: bound ],
        newListener: [Function: bound ],
        stateChange: [Function],
        discover: [AsyncFunction] },
     _eventsCount: 4,
     _allowDuplicates: undefined },
  id: '3b298bc874144c2cad1bcac3905ed5c0',
  uuid: '3b298bc874144c2cad1bcac3905ed5c0',
  address: '',
  addressType: 'unknown',
  connectable: true,
  advertisement:
   { manufacturerData: <Buffer 57 01 00 29 1c 9c 59 10 7c b5 fc eb 3a fa fe 39 48 1d fc 01 d0 46 82 41 5a 79> },
  rssi: -87,
  services: null,
  state: 'disconnected' }`

geovie, 2018-12-14:
@ReinierH Yes that's expected, apparently they are not delivered by the OS on every discover event. #17 changed the behavior so that we only update advertisement properties which have changed, as noble would override the previous property if we set the property to a default value (see also #16 ).
So the advertisement will be updated once we get the advertisement info from the OS.

Maybe wait a few seconds and check the device then...

s524797336, 2018-12-18:
@geovie  i tested new 0.0.3 and everything work perfect

geovie, 2018-12-18:
@s524797336 Great, thanks for testing!

Fixed in `0.0.3`

@jaret32 @ReinierH I'll close this issue, if you're still having problems please open a new one, thank you!

closed 2018-12-18 by geovie