Timeular/noble-mac #11: app crashes constantly when stateChange is attached after require

tzvetelin-vassilev, 2018-11-07:
I have electron environment. App not starting successfully very often. I have isolated the following code:

```javascript
const ble = require("noble-mac");
ble.on("stateChange", (state) => console.log("bleState", state));
```

geovie, 2018-11-07:
I can't reproduce it with this snippet, which electron version are you using. Can you provide a small electron sample? Also which MacOS version are you using?

tzvetelin-vassilev, 2018-11-08:
Electron is v3.0.8, node is v11.0.0, MacOS is Mojave 10.14. xpc-connection is sandeepmistry fix for node higher then 10.

  "dependencies": {
    "xpc-connection": "https://github.com/sandeepmistry/node-xpc-connection#pull/26/head",
    "noble-mac": "git+https://github.com/Timeular/noble-mac.git",
  }

With very simple app it happens very rarely. In fact I reproduced it only once out of real app context.
In the beginning state is unknown. If the event is attached after first state change everything is OK. Next changes works as expected.

geovie, 2018-11-08:
@tzvetelin-vassilev Please use this [branch](https://github.com/Timeular/noble-mac/compare/test/app-crash), it builds in debug mode this way the back-trace will be more useful. 
e.g:
```json
"dependencies": {
  "noble-mac": "git+https://github.com/Timeular/noble-mac#e84efde7cf71a7fc022dc79e8cc3e316b08b2154"
}
```

tzvetelin-vassilev, 2018-11-08:
[invalid-access2.txt](https://github.com/Timeular/noble-mac/files/2562361/invalid-access2.txt)

geovie, 2018-11-09:
@tzvetelin-vassilev 🤔 that didn't work.. Are you sure that you used the right branch? Can you debug your app with `lldb`?

tzvetelin-vassilev, 2018-11-09:
I don't know how, but if you give me some instructions what to do?

geovie, 2018-11-09:
Open a terminal and type `lldb <your app with args>` (e.g. `lldb electron main.js`) and then type `run` and return key to start it. After the crash you can type `bt` to get a back trace...

tzvetelin-vassilev, 2018-11-09:
lldb electron main.js                                                                                                                                                                                                                                                                                         
(lldb) target create "electron"
error: '/usr/local/bin/electron' doesn't contain any 'host' platform architectures: x86_64h, x86_64, i386, x86_64
some help here?

geovie, 2018-11-09:
Try it with `./node_modules/.bin/electron` in your project root

tzvetelin-vassilev, 2018-11-09:
the same error
can you put some logs to the console in this branch to allocate where problem appears?

tzvetelin-vassilev, 2018-11-09:
[invalid-access3.txt](https://github.com/Timeular/noble-mac/files/2566960/invalid-access3.txt)

geovie, 2018-11-12:
@tzvetelin-vassilev Thanks for the stack trace, I think I've found the problem.
Could you try the latest commit of the [test branch](https://github.com/Timeular/noble-mac/tree/test/app-crash)? This should fix the crash.

tzvetelin-vassilev, 2018-11-12:
I'm not sure how exactly to refer it. git+https://github.com/Timeular/noble-mac#test or probably something else?

geovie, 2018-11-12:
@tzvetelin-vassilev Try the commit hash: `git+https://github.com/Timeular/noble-mac#38ac14b65cef7d8753724415db0543440efff93e`

tzvetelin-vassilev, 2018-11-12:
yes, it is fixed, there is no crash any more

geovie, 2018-11-12:
Thanks for testing!