Timeular/noble-mac #13: Still can't get it to run on 10.14

watsonix, 2018-11-09:
I've attempted to use this solution with https://github.com/chbrown/BluetoothLE-HeartRate
however after I change my dependencies to:

`  "dependencies": {
    "noble-mac": "Timeular/noble-mac"
  }` 

and my top line of `index.js` to: `const noble = require('noble-mac')`

I still can't get it to run: 

```
$ node index.js 
internal/modules/cjs/loader.js:589
    throw err;
    ^

Error: Cannot find module '../native/noble_mac'
```

I've tried installing and re-installing node a million times, installing globally and all manner of other things.

geovie, 2018-11-09:
@watsonix Do you get any error during `npm install`/`yarn install`? Which node version are you using? Can you post your project somewhere?

watsonix, 2018-11-09:
@geovie thanks for asking. I've forked the project and put my changes up here:
https://github.com/watsonix/BluetoothLE-HeartRate

Here's the diff:
```
$ git diff
diff --git a/index.js b/index.js
index ae30807..8c73632 100644
--- a/index.js
+++ b/index.js
@@ -1,4 +1,4 @@
-const noble = require('noble')
+const noble = require('noble-mac')
 
 /**
 Ensure that noble is ready, then scan for peripherals implementing the given service UUIDs.
diff --git a/package.json b/package.json
index df88811..91a9a32 100644
--- a/package.json
+++ b/package.json
@@ -12,6 +12,6 @@
   "author": "Christopher Brown <io@henrian.com>",
   "license": "MIT",
   "dependencies": {
-    "noble": "^1.8.1"
+    "noble-mac": "Timeular/noble-mac"
   }
 }

```

Here's what I see when I try to install ([...] is my deletion/annotation):
```
$ node --version
v10.13.0
MacBook-Air:BluetoothLE-HeartRate watson$ npm --version
6.4.1
MacBook-Air:BluetoothLE-HeartRate watson$ npm install
npm ERR! path /Users/watson/.npm/_cacache/tmp/git-clone-0536e269
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall mkdir
npm ERR! Error: EACCES: permission denied, mkdir '/Users/watson/.npm/_cacache/tmp/git-clone-0536e269'
[...]
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

MacBook-Air:BluetoothLE-HeartRate watson$ sudo npm install
[...]
gyp ERR! stack Error: EACCES: permission denied, mkdir '/Users/watson/Dropbox/code/core/ble_hr/BluetoothLE-HeartRate/node_modules/xpc-connection/build'
[...]

[also tried: sudo npm install --unsafe-perm=true --allow-root]

MacBook-Air:BluetoothLE-HeartRate watson$ node index.js 
internal/modules/cjs/loader.js:582
    throw err;
    ^

Error: Cannot find module '../native/noble_mac'
[...]
```

geovie, 2018-11-09:
@watsonix You shouldn't have to use `sudo` with `npm install`!
I tested your project and it works fine for me. This appears to be an issue with your settings (`EACCES: permission denied`) which is unrelated to noble-mac.

closed 2018-11-09 by geovie

watsonix, 2018-11-10:
Hmmm, am I doing this correctly? 
```
   "dependencies": {
       "noble-mac": "Timeular/noble-mac"
   }
```

watsonix, 2018-11-11:
I reinstalled Node with https://github.com/brock/node-reinstall/ and now my fork is working fine and I'm getting data