Timeular/noble-mac #15: Can't get address by connecting and disconnecting to device

vikeri, 2018-12-10:
## Expected behavior:
When finding a new device (that will lack the address) I can connect and then disconnect. Then the device will show up in the scan and it will contain the address.

## Actual behavior:
Doesn't matter if I disconnect the device still has no address when found again. Has to restart the whole script in order for it to find the address.

## Long version:
I have a script that finds a device based on name. If the device didn't have the address I only had to connect to the device and then disconnect and the next time the device was found in the scan the address was available. But now it doesn't matter if I disconnect and then find the device again, it will still not find the address.

geovie, 2018-12-10:
Hi, I think this is a problem with noble, as you can see [here](https://github.com/noble/noble/blob/master/lib/noble.js#L149) the address is only updated if the peripheral is not yet discovered.
You say that it worked before:
> ... If the device didn't have the address I only had to connect to the device and then disconnect and the next time the device was found in the scan the address was available. But now it doesn't matter if I disconnect and then find the device again, it will still not find the address.

Which noble/noble-mac version where you using?

vikeri, 2018-12-12:
Interesting, I'm using 0.0.2

geovie, 2018-12-12:
And where you using noble before (when it worked) or was that another module?

geovie, 2018-12-18:
@vikeri I'll close this, as it is most likely the problem of how noble handles the device address. If you feel this is still an issue with `noble-mac` please reopen this or create a new issue, thanks!

closed 2018-12-18 by geovie