Timeular/noble-mac #16: 'advertisement' object is being released

gpgabriel, 2018-12-11:
Hi, I get some strange behaviour and I'm trying to find the root issue.

I created a script to reproduce this bug:
https://gist.github.com/gpgabriel/2007be8f27ea79189b19a5e35ed0ad1a

The bug only happens on macOS 10.14.1, I tried it with node 8.14.0.

```
Pushed device:  MY_BLE_DEVICE
Devices found:  1 [ 'MY_BLE_DEVICE' ]
Devices found:  1 [ 'MY_BLE_DEVICE' ]
Devices found:  1 [ 'MY_BLE_DEVICE' ]
Devices found:  1 [ 'MY_BLE_DEVICE' ]
Pushed device:  MY_BLE_DEVICE
Devices found:  2 [ 'MY_BLE_DEVICE', 'MY_BLE_DEVICE' ]
Devices found:  2 [ 'MY_BLE_DEVICE', 'MY_BLE_DEVICE' ]
Devices found:  2 [ 'MY_BLE_DEVICE', 'MY_BLE_DEVICE' ]
Devices found:  2 [ 'MY_BLE_DEVICE', 'MY_BLE_DEVICE' ]
Pushed device:  MY_BLE_DEVICE
Stop scanning
```

```
Pushed device:  MY_BLE_DEVICE
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Stop scanning
```

```
Pushed device:  MY_BLE_DEVICE
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Devices found:  1 [ '' ]
Pushed device:  MY_BLE_DEVICE
Devices found:  2 [ 'MY_BLE_DEVICE', 'MY_BLE_DEVICE' ]
Devices found:  2 [ 'MY_BLE_DEVICE', 'MY_BLE_DEVICE' ]
Stop scanning
```

The problem is that the `advertisement` property from the peripheral object is reseted to default values. It's still there but it's reseted to:

```
{
  localName: '',
  txPowerLevel: -1680765016,
  manufacturerData: <Buffer >,
  serviceData: [],
  serviceUuids: []
}
```

Any idea how to avoid this?

Thanks,
Gabriel

gpgabriel, 2018-12-11:
The results are never consistent, sometimes the advertisement object stays untouched and sometimes is being reseted.

geovie, 2018-12-11:
Hi, we're currently not handling the advertisement correctly. As you can see [here](
https://github.com/noble/noble/blob/c4cd18a7a429bb832f6c4ca793be3faf9af884e9/lib/noble.js#L160-L165) noble expects advertisement properties to be `undefined` for them to not be updated, which does not fit well with the cpp model (although maybe std::optional could be used...).
The first thing to check is if there is no data (nil) in the `CBAdvertisementDataLocalNameKey` advertisement or if it is empty. Could you check this by logging the peripheral name (`p.name`) in [`ble_manager.mm:55`](
https://github.com/Timeular/noble-mac/blob/master/src/ble_manager.mm#L55)?

gpgabriel, 2018-12-11:
@geovie I don't have any experience with Objective-C, I tried to use `NSLog(@"%@", p.name);` but it throws an error. How can I log the `p.name` variable?

geovie, 2018-12-11:
@gpgabriel you can use `printf("AdvName: %s\n", p.name.c_str());` or you can also put a `NSLog(@"%@", [advertisementData objectForKey:CBAdvertisementDataLocalNameKey]);` before the if.

gpgabriel, 2018-12-11:
I found it 1 second before you posted `printf("ble_manager.mm: %s\n", p.name.c_str());`

gpgabriel, 2018-12-11:
```
ble_manager.mm: Charge 2
ble_manager.mm: MY_BLE_DEVICE
Pushed device:  MY_BLE_DEVICE
Stop scanning
[ Peripheral {
    _noble: 
     Noble {
       initialized: true,
       address: 'unknown',
       _state: 'poweredOn',
       _bindings: [Object],
       _peripherals: [Object],
       _services: [Object],
       _characteristics: [Object],
       _descriptors: [Object],
       _discoveredPeripheralUUids: [Array],
       _events: [Object],
       _eventsCount: 4,
       _allowDuplicates: true },
    id: '1bd0a0655f7e4212b096af67623331f8',
    uuid: '1bd0a0655f7e4212b096af67623331f8',
    address: '',
    addressType: 'unknown',
    connectable: false,
    advertisement: 
     { localName: '',
       txPowerLevel: -1680765016,
       manufacturerData: <Buffer >,
       serviceData: [],
       serviceUuids: [] },
    rssi: -72,
    services: null,
    state: 'disconnected' } ]
```

gpgabriel, 2018-12-11:
I know that Objective-C is not garbage collected but is there any way the `Peripheral p` variable is freed after it was emitted?

geovie, 2018-12-11:
🤔 Can you also log the name [here](https://github.com/Timeular/noble-mac/blob/master/src/callbacks.cc#L83) and [here](https://github.com/Timeular/noble-mac/blob/master/src/callbacks.cc#L90)?

gpgabriel, 2018-12-11:
Here is the result. https://gist.github.com/gpgabriel/9870d23ff5f411fbfa26740b4616e225

```
// line 84
printf("callbacks.cc:83 %s\n", name.c_str());

// line 91
printf("callbacks.cc:90 %s\n", name.c_str());
```

geovie, 2018-12-11:
Please also log the `uuid` in callbacks.cc to identify `MY_BLE_DEVICE`

gpgabriel, 2018-12-11:
Hope this is useful https://gist.github.com/gpgabriel/47e98dd15767ceb77dae118cd646d23a

geovie, 2018-12-11:
As you can see the uuid of your device appears two times: first with the correct name `MY_BLE_DEVICE` and then with an empty name (no log from blemanager.cc which means `[advertisementData objectForKey:CBAdvertisementDataLocalNameKey]` is `nil`)

A hacky solution would be to ignore empty local names when emitting a peropheral:
```cpp
if (!name.empty()) {
  advertisment.Set(_s("localName"), _s(name));
}
```
in this [line](https://github.com/Timeular/noble-mac/blob/master/src/callbacks.cc#L90).

The correct solution would be to put the `name` field (maybe also other fields) of [`Peripheral`](https://github.com/Timeular/noble-mac/blob/master/src/peripheral.h#L18) in a [std::optional](https://en.cppreference.com/w/cpp/utility/optional) and only call `advertisment.Set` if the optional is set.

I won't manage to change this anytime soon, but I could guide you (or some one else) if you want to give it a try...

gpgabriel, 2018-12-11:
Thanks for looking at this issue, I'll try to open a PR for this.

gpgabriel, 2018-12-11:
`std::optional` is only available in libc++ C++17 which is not complete yet. It sounds a bit risky to use it now?

geovie, 2018-12-11:
Yeah too bad... It mentions `std::pair<T,bool>` (so `std::pair<std::string,bool>`) as alternative. Or you could create a custom class:
```cpp
template <typename T>
class Option {
private:
    T value;
    bool set;
public:
    Option() : set(false) {
    }
    Option(const T& val) : set(true), value(val) {
    }

   bool isSet() {
     return set;
   }
}
```

gpgabriel, 2018-12-12:
Thanks @geovie! The PR was merged.

closed 2018-12-12 by gpgabriel