Timeular/noble-mac #18: Error on Electron

itome, 2018-12-16:
I'm using this library for electron app.
And I got
```
FATAL ERROR:  Cannot read property '0' of undefined
 1: node::Abort() [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
 2: node::FatalError(char const*, char const*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
 3: node::FatalError(char const*, char const*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
 4: napi_fatal_error [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
 5: Napi::Error::Fatal(char const*, char const*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/noble-mac/native/noble_mac.node]
 6: ThreadSafeCallback::Impl::static_async_callback(uv_async_s*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/noble-mac/native/noble_mac.node]
 7: uv_async_send [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
 8: uv_free_interface_addresses [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
 9: uv_run [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libnode.dylib]
10: atom::NodeBindings::UvRunOnce() [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
11: base::internal::Invoker<base::internal::BindState<void (atom::api::BrowserWindow::*)(), base::WeakPtr<atom::api::BrowserWindow> >, void ()>::Run(base::internal::BindStateBase*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
12: base::internal::Invoker<base::internal::BindState<void (atom::api::BrowserWindow::*)(), base::WeakPtr<atom::api::BrowserWindow> >, void ()>::Run(base::internal::BindStateBase*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
13: base::internal::Invoker<base::internal::BindState<void (atom::api::BrowserWindow::*)(), base::WeakPtr<atom::api::BrowserWindow> >, void ()>::Run(base::internal::BindStateBase*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
14: base::internal::Invoker<base::internal::BindState<void (atom::NodeBindings::*)(), base::WeakPtr<atom::NodeBindings> >, void ()>::RunOnce(base::internal::BindStateBase*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
15: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
16: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
17: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
18: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
19: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
20: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
21: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
22: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
23: __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ [/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation]
24: __CFRunLoopDoSource0 [/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation]
25: __CFRunLoopDoSources0 [/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation]
26: __CFRunLoopRun [/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation]
27: CFRunLoopRunSpecific [/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation]
28: -[NSRunLoop(NSRunLoop) runMode:beforeDate:] [/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation]
29: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
30: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
31: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
32: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
33: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
34: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
35: crashpad::internal::ScopedDIRCloseTraits::Free(DIR*) [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
36: AtomMain [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework]
37: main [/Users/takeshi/Projects/mabeee-desktop/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Helper.app/Contents/MacOS/Electron Helper]
38: start [/usr/lib/system/libdyld.dylib]
```
I thisnk this may be caused by native modules binding. Any idea about this error?

s524797336, 2018-12-18:
I'm using this too, no error found, what electron version are you using?
electron v3.0.12
node v10.10.0

geovie, 2018-12-18:
@itome I think the error means that you're trying to access a property of `undefined` inside a event handler. Could you post a snippet of your code?

itome, 2018-12-19:
Installation succeeded by rebuilding project dependencies. But I don't know why the exception resolved.

Now I can discover devices. But error appears when connecting to devices. 

Error Message:
```
2018-12-20 00:35:34.027 Electron Helper[17666:6766651] *** Assertion failure in -[CBPeripheral writeValue:forDescriptor:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/MobileBluetooth/MobileBluetooth-121.13/CoreBluetooth/CoreBluetooth/CBPeripheral.m:533
2018-12-20 00:35:34.057 Electron Helper[17666:6766651] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff44ec7e65 __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff70f23720 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff44ee2ab2 +[NSException raise:format:arguments:] + 98
	3   Foundation                          0x00007fff47286d1d -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 194
	4   CoreBluetooth                       0x00007fff4492dd4b -[CBPeripheral writeValue:forDescriptor:] + 1111
	5   noble_mac.node                      0x0000000111633de6 -[BLEManager writeValue:service:characteristic:descriptor:data:] + 240
	6   noble_mac.node                      0x000000011162bb96 _ZN8NobleMac10WriteValueERKN4Napi12CallbackInfoE + 1116
	7   noble_mac.node                      0x000000011162d717 _ZZN4Napi10ObjectWrapI8NobleMacE29InstanceMethodCallbackWrapperEP10napi_env__P20napi_callback_info__ENKUlvE_clEv + 129
	8   noble_mac.node                      0x000000011162d653 _ZN4Napi10ObjectWrapI8NobleMacE29InstanceMethodCallbackWrapperEP10napi_env__P20napi_callback_info__ + 39
	9   libnode.dylib                       0x000000010c0741e0 napi_run_script + 944
	10  libnode.dylib                       0x000000010c06da53 napi_create_function + 755
	11  libnode.dylib                       0x000000010c5c7dd4 _ZN2v88internal13VirtualMemory11TakeControlEPS1_ + 1172
	12  libnode.dylib                       0x000000010c61a352 _ZN2v88internallsERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEERKNS0_18BasicBlockProfilerE + 138194
	13  libnode.dylib                       0x000000010c619a3c _ZN2v88internallsERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEERKNS0_18BasicBlockProfilerE + 135868
	14  ???                                 0x00000044f130453d 0x0 + 296104248637
)
```

itome, 2018-12-21:
Sorry, I understand descriptor limitation and that error resolved.
But I met first error again.

geovie, 2018-12-21:
@itome My guess is that this is because something inside a callback is throwing e.g.:
```javascript
peripheral.connect(function(error) {
  throw new Error(); // this will cause an error similar to the above
});
```
So this is a problem with your code, if you can share it I can take a look (most likely you're try to access an array which is undefined)

itome, 2018-12-21:
Thank you. I try to access serviceData[0] when peripheral discovered, and it throw an error.
It's my mistake. Sorry. I close this issue.

closed 2018-12-21 by itome