Timeular/noble-mac #19: installation question

ainthek, 2018-12-19:
Please is this installable from npm somewhere ?
or just from git ?

Any chance this will be merged with noble project or is the noble completely abandoned ?

bankyadam, 2018-12-26:
@geovie Did you forget to publish 0.0.3 to npm?

chrkhl, 2019-01-06:
Can someone please publish release 0.0.3 to npm? Thank you!

jongear, 2019-01-17:
since the package isn't on npm I have been referencing it directly from git.

```json
"dependencies": {
    "noble-mac": "https://github.com/Timeular/noble-mac.git"
  },
```

chrkhl, 2019-01-17:
> since the package isn't on npm I have been referencing it directly from git.
> 
> ```json
> "dependencies": {
>     "noble-mac": "https://github.com/Timeular/noble-mac.git"
>   },
> ```

Yes, I do the same, but I hope that there will be an npm package after all.

keeslinp, 2019-01-18:
Are there any plans for fixing this? We're using this as a fix for a project using noble that needs to work on mojave, but part of our build process is unhappy with git dependencies. Changing this to npm would solve that problem.

s524797336, 2019-02-11:
@keeslinp why not fork this project and publish a private npm package instead.

geovie, 2019-11-07:
FYI: noble-mac is now included in https://github.com/abandonware/noble

closed 2019-11-07 by geovie