Timeular/noble-mac #20: this._bindings.write - There should be 5 arguments: (String, String, String, Buffer, Boolean)

alexmerz, 2019-01-09:
Hi,

i'm using the current github version of noble-mac on MacOS High Sierra with Node 8.9.4. 

When I try to write to an characteristic, I get this error:
```
/Users/alexandermerz/github/dab_ws1819/blcentral/node_modules/noble/lib/noble.js:325
   this._bindings.write(peripheralUuid, serviceUuid, characteristicUuid, data, withoutResponse);
                  ^

TypeError: There should be 5 arguments: (String, String, String, Buffer, Boolean)
    at Noble.write (/Users/alexandermerz/github/dab_ws1819/blcentral/node_modules/noble/lib/noble.js:325:19)
    at Characteristic.write (/Users/alexandermerz/github/dab_ws1819/blcentral/node_modules/noble/lib/characteristic.js:74:15)
    at Timeout.executeCommand [as _onTimeout] (/Users/alexandermerz/github/dab_ws1819/blcentral/app.js:48:19)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
```

A look into the code of noble&noble-mac shows no difference in the number of function arguments. The noble-mac library/executable was installed by node-pre-gyp via download.

Any idea whats going wrong here?

geovie, 2019-01-09:
Ensure that the arguments are of correct type, I guess the data you're giving is maybe not a Buffer...
Can you post your code?

geovie, 2019-01-09:
@alexmerz You're missing the `withoutResponse` (bool) argument (https://github.com/noble/noble#write)

alexmerz, 2019-01-09:
@geovie You are right. Sorry for that stupid mistake. Thank you!

closed 2019-01-09 by alexmerz