Timeular/noble-mac #21: Advertisement.ManufacturerData comes back as buffer

jongear, 2019-01-17:
OS: Mojave 10.14.2
Node: 10

When attempting to access the manufacturer data of the peripheral `noble-mac` exposes a buffer. 

[image]

stringify and utf-8 conversions return 

```
{"type":"Buffer","data":[67,0,207,25,11,9,1,0]}
<Buffer 43 00 cf 19 0b 09 01 00>
C�
```

geovie, 2019-01-18:
That's expected. The manufacturerData is a Buffer as defined in: https://github.com/noble/noble/blob/master/README.md#peripheral-discovered 
Is the Data truncated?

jongear, 2019-01-18:
You're correct. The data is coming back as expected and is not truncated. I was not casting properly. To find my manufacturer. In my case I neeeded `.toString('hex')`

[image]

closed 2019-01-18 by jongear