Timeular/noble-mac #22: Typeerror when using `Peripheral.readHandle`

tal, 2019-01-23:
When I try and call `Peripheral.readHandle` I get the following error:

```
TypeError: There should be 4 arguments: (String, String, String, String)
 at Noble.readHandle (/Users/atlastal/Projects/bt-explore/node_modules/noble/lib/noble.js:429:18)
 at Peripheral.readHandle (/Users/atlastal/Projects/bt-explore/node_modules/noble/lib/peripheral.js:120:15)
 at p.connect (/Users/atlastal/Projects/bt-explore/dist/index.js:15:19)
 at Peripheral.<anonymous> (/Users/atlastal/Projects/bt-explore/node_modules/noble/lib/peripheral.js:38:7)
 at Object.onceWrapper (events.js:315:30)
 at emitOne (events.js:116:13)
 at Peripheral.emit (events.js:211:7)
 at Noble.onConnect (/Users/atlastal/Projects/bt-explore/node_modules/noble/lib/noble.js:191:16)
 at emitTwo (events.js:126:13)
 at NobleMac.emit (events.js:214:7)
```

tal, 2019-01-23:
ah sorry, his is an issue with https://github.com/noble/noble

tal, 2019-01-23:
Actually I take it back. it appears as though it's an issue with the handle. Not sure if that's the mac or root noble. Any thoughts?

https://github.com/noble/noble/blob/master/lib/noble.js#L428-L430

geovie, 2019-01-23:
This is a bug in noble-mac, I will fix it soon.
Thanks for reporting!

geovie, 2019-02-15:
Fixed, Thanks

PR #29 (2019-02-15): Fix readHandle/writeHandle, rename cleanUp to stop
Fixes #22 and #26
commit 578b723d (2019-02-15): Fix readHandle/writeHandle, rename cleanUp to stop (#29)

Fixes #22 and #26
closed 2019-02-15 by geovie