Timeular/noble-mac #23: Node 10 support on linux

hannseman, 2019-01-26:
I understand that the main purpose of this package is to bring support for macOS Mojave but I would like to ask you if you could consider depending on https://github.com/abandonware/noble which fixes Node 10 support for linux. This is the root issue https://github.com/noble/node-bluetooth-hci-socket/issues/84.

This would enable people who have to support both macOS Mojave and Node >=10 on Linux to use this library.

hannseman, 2019-02-07:
@geovie looks like `abandonware/noble` is interested in integrating. Please see https://github.com/abandonware/noble/issues/4. 

That fork is pretty popular with 1600 weekly downloads.
https://www.npmjs.com/package/@abandonware/noble

Compare that to the original package with has 1800 weekly downloads.
https://www.npmjs.com/package/noble

rzr, 2019-08-06:
Are there any patches from this fork that are pending upstream ?

If yes they could be merged into @abandonware/noble

Again efforts to track changes are appreciated:
https://github.com/abandonware/noble/issues/2

geovie, 2019-08-06:
@rzr No @abandonware/noble already contains the latest changes https://github.com/abandonware/noble/pull/8

rzr, 2019-08-06:
Feel free to rebase on @abandonware/noble if you want, I plan to keep it based on upstream too