Timeular/noble-mac #26: Is the CleanUp method the same thing as you'd expect stop to be/do?

jrobeson, 2019-02-12:
I'm thinking about implementing a way to start and stop the bindings, so I just wondered if you considered `CleanUp` to be equivalent to that.

geovie, 2019-02-12:
Yes, `cleanUp` should release any native resources that are held and allows the noble (binding) instance to be GC'ed.  I'm not sure if it's already well tested, I know it works for the simple case, but I'm not sure what happens if you cleanup a binding which has a device connected.

jrobeson, 2019-02-12:
So should it be renamed to `stop` to match bluetooth-hci-socket?

geovie, 2019-02-13:
Sure I can rename it. Any other renames or additional function that would make sense? Which common interface do you have in mind?

jrobeson, 2019-02-13:
Nothing  else ATM. I just saw an old noble bug re: missing stop method, and it mentioned that the bindings would need to have it first before it could be exposed at the noble level.

geovie, 2019-02-15:
Renamed, I will create a new build soon.

closed 2019-02-15 by geovie