Timeular/noble-mac #27: Unable to Set ArrayBuffer Value for Descriptor

drewcovi, 2019-02-12:
Attempting to set the following descriptors fails on the second descriptor. This approach works flawlessly on Linux, assuming something is getting lost in translation for the second descriptors arraybuffer

```js
descriptors: [
                new Descriptor({
                    uuid: '2901',
                    value: Buffer.from('Volume', 'utf8')
                }),
                new Descriptor({
                    uuid: '2904', // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_presentation_format.xml
                    value: new Buffer([
                        0x04, // Uint8 0-255 value is allowed (numeric keypad)
                        0x00, // no exponent (value === value)
                        0x27, 0x25, // https://www.bluetooth.com/specifications/assigned-numbers/units joules (energy)
                        0x01, // Bluetooth SIG Assigned Numbers
                        0x00, // No description
                        0x00])
                })
            ]
```

```
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Expecting NSData value type for Format descriptor'
```

geovie, 2019-02-13:
Could you provide a small code snippet which reproduces this?

drewcovi, 2019-02-13:
gah! i mixed up bleno and noble again :) sorry.

closed 2019-02-13 by drewcovi