Timeular/noble-mac #30: Fatal error on NodeV8 OSX

allnash, 2019-03-04:
Installed the latest version with Node V8. It blew up!

```FATAL ERROR:  Cannot read property 'find' of undefined
 1: node::Abort() [/usr/local/bin/node]
 2: node::FatalException(v8::Isolate*, v8::Local<v8::Value>, v8::Local<v8::Message>) [/usr/local/bin/node]
 3: node::OnFatalError(char const*, char const*) [/usr/local/bin/node]
 4: napi_fatal_error [/usr/local/bin/node]
 5: Napi::Error::Fatal(char const*, char const*) [/Users/gadre/ciign/device/scanner/node_modules/noble-mac/native/noble_mac.node]
 6: ThreadSafeCallback::Impl::static_async_callback(uv_async_s*) [/Users/gadre/ciign/device/scanner/node_modules/noble-mac/native/noble_mac.node]
 7: uv__async_io [/usr/local/bin/node]
 8: uv__io_poll [/usr/local/bin/node]
 9: uv_run [/usr/local/bin/node]
10: node::Start(v8::Isolate*, node::IsolateData*, int, char const* const*, int, char const* const*) [/usr/local/bin/node]
11: node::Start(uv_loop_s*, int, char const* const*, int, char const* const*) [/usr/local/bin/node]
12: node::Start(int, char**) [/usr/local/bin/node]
13: start [/usr/lib/system/libdyld.dylib]

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

geovie, 2019-03-04:
Could you provide a code snippet which leads to the error, thanks!

geovie, 2019-03-12:
@allnash Were you able to find the cause of the crash?

allnash, 2019-03-13:
I think this maybe caused by bleno, I did not look further.

geovie, 2019-03-13:
@allnash Ok, then I'll close this, feel free to reopen

closed 2019-03-13 by geovie