Timeular/noble-mac #32: Write to characteristic not working on Mac

elafargue, 2019-04-07:
On Mojave, writing to characteristics doesn't seem to work at all - the same exact code on a Linux machine writes fine, but on the Mac it looks like event though there is no error, data doesn't make it out to the peripheral.

Has characteristic writing been tested on this new MacOS native implementation?

geovie, 2019-04-07:
@elafargue Worked so far for me. Can you post a snippet of your code? Are you sure that you call write with the correct arguments? Maybe have a look at https://github.com/Timeular/noble-mac/issues/20

elafargue, 2019-04-07:
This is the write method in my program - it has been working for ages (>1 year, used on a bunch of stations connected to BLE sensors out there), but stopped for the Mac when migrating to Mojave, which led me to try this fork. Can you see something that looks incorrect there?

`        this.write = function (data, info, callback) {
            if (!portOpen || info == undefined || callback == undefined)
                return;
            var s = findService(info.service_uuid);
            s.discoverCharacteristics([ info.characteristic_uuid.replace(/-/gi,'') ], function(err, c) {
                if (typeof c == 'undefined' ) {
                    debug('Error: could not find a matching characteristic');
                    self.emit('status', {
                        openerror: true,
                        reason: 'Could not connect to the BLE service',
                        description: 'Did not find characteristic.'
                    });
                    self.close();
                    return;
                }

                //  Now we can write
                var dab = Buffer.from(data);
                var noresp = c[0].properties.indexOf('writeWithoutResponse') != -1;
                c[0].write(dab, noresp, function(err) {debug('writing result', err)});
            });
        };
`

buttjer, 2019-04-09:
> This is the write method in my program - it has been working for ages (>1 year, used on a bunch of stations connected to BLE sensors out there), but stopped for the Mac when migrating to Mojave, [...]

Not sure if this helps but writeWithoutResponse is unused in noble-mac. You 'll get response bothway.