Timeular/noble-mac #34: Support for node version 10.15.3

lilliesAndRoses, 2019-04-25:
Is there any plan of adding support for node version 10.15.3?

geovie, 2019-04-25:
@lilliesAndRoses Is it not working for you with 10.15.3?

lilliesAndRoses, 2019-04-26:
I am getting following error while installing.

```
~/Documents/app :npm install noble-mac
npm WARN npm npm does not support Node.js v10.15.3
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
npm ERR! code E404
npm ERR! 404 Not Found: noble-mac@latest
```

geovie, 2019-04-26:
@lilliesAndRoses I think you need to update your `npm` also be sure to add `noble-mac` as [git dependency](https://github.com/Timeular/noble-mac#installation) as it is not on npm.
You might want to try the [abandonware fork](https://github.com/abandonware/noble/) of noble, which contains `noble-mac`.

s524797336, 2019-04-30:
@lilliesAndRoses 
Your current npm not support current node version.
Use `npm i -g npm` to update npm, you may need `sudo` for that.

lilliesAndRoses, 2019-05-02:
Thanks. Adding a  'git dependency' worked. 
Closing the issue.

closed 2019-05-02 by lilliesAndRoses