Timeular/noble-mac #35: Moving forward

floating, 2019-05-18:
Hi all, 

I want to start out by saying thanks for this project. I'm wondering what the plan is moving forward. Do you think this should be merged back into noble or do you think it should be standalone? If standalone do you think we can have an official release on npm? As someone who needs to support newer macOS platforms I now rely on this library now instead of noble.

Thanks again!
Jordan

s524797336, 2019-05-20:
There are no one maintained noble now. so...

geovie, 2019-05-23:
@floating `noble-mac` is now included in [abandonware noble fork](https://github.com/abandonware/noble)    
Checkout [@abandonware/noble](https://www.npmjs.com/package/@abandonware/noble)

closed 2019-11-07 by geovie