Timeular/noble-mac #36: Question: Documentation for adapters

LukasBombach, 2019-07-12:
Hey,

I am trying to rewrite parts of Noble in TypeScript and I am looking for some documentation on adapters. Can you tell me how you figured out how to implement your adapter (interface)?

Cheers
Lukas

geovie, 2019-07-17:
@LukasBombach Hi, there is not much documentation except the code it self.   
You could take a look at [noble](https://github.com/noble/noble/blob/master/lib/noble.js)(basically everything that uses `_bindings`) or [ble manager](https://github.com/Timeular/noble-mac/blob/master/src/ble_manager.h).    
jrobeson's [typescript refactor](https://github.com/jrobeson/noble/tree/refactor) fork might also have some infos

closed 2019-07-17 by geovie

LukasBombach, 2019-07-17:
Cool thanks; I ended up using https://github.com/noble/noble/blob/master/lib/websocket/bindings.js as a reference