Timeular/noble-mac #37: cannot find module '../native/noble_mac'

slayer1021, 2019-09-22:
Followed instructions for installation but keep getting the error   cannot find module '../native/noble_mac' , did check the location and no file exists in the location.

kenken64, 2019-12-20:
Error: Cannot find module '../native/noble_mac'

geovie, 2019-12-20:
@slayer1021 @kenken64 Which node version are you using? How are you installing `noble-mac`?
You might want to take a look at [abandonware/noble](https://github.com/abandonware/noble) which has the bindings of this repo integrated.

blamay, 2021-04-12:
I experienced this issue on both this repo & on abandonware/noble, and it was because I am using webpack which doesn't play nicely with the `binding.node` file, or in this repos case, the 'noble_mac.node' file. I was able to fix it with https://github.com/webpack-contrib/node-loader but it required me making this change to `binding.js`

> const NobleMac = require('node-loader!./native/binding.node').NobleMac;

Unfortunately I have to 'fork' abandonware to make this change. I think there could potentially be a change to allow one to do this and pass it along as an option. It would be nice to be able to keep `noble` in node_modules.

geovie, 2021-04-13:
@blamay Could you specify `noble-mac` as [external](https://webpack.js.org/configuration/externals/) in webpack, then it should work...

blamay, 2021-05-04:
Specifying noble (or in my case, @abandonware/noble) as external does fix this problem for me.

ajjimsa, 2022-11-23:
@blamay Possibly you don't have  added in webpack file  to load the .node extensions.
...
 resolve: {
    extensions: [".js", ".css",".node","..."],
  },
...