Timeular/noble-mac #41: Native Addon does not seem to terminate

LukasBombach, 2019-12-26:
Hey there

if I load the compiled native node addon directly and run this little script:

```js
const util = require("util");
const { EventEmitter } = require("events");
const NobleMac = require("./native/noble_mac").NobleMac;

util.inherits(NobleMac, EventEmitter);

const nobleMac = new NobleMac();

nobleMac.init();

console.log("done");
```

The node process will never terminate. If you remove the line `nobleMac.init();` it will terminate right after logging `done`. For some reason the native process seems to keep running and I cannot find the reason nor how to terminate it.

Do you have an idea? <3

LukasBombach, 2019-12-26:
Ok it is this line

https://github.com/Timeular/noble-mac/blob/master/src/ble_manager.mm#L18

Maybe I can find a fix and open a PR

geovie, 2019-12-26:
@LukasBombach calling the stop method should work https://github.com/Timeular/noble-mac/blob/master/src/noble_mac.mm#L249

LukasBombach, 2019-12-27:
OH YES

Thank you soo much! 🎉

closed 2019-12-27 by LukasBombach